LISE-B26/pylabcontrol #111: Settings don't seem to always update properly in the instrument tab

akabcenell, 2017-01-11:
See for example MagnetCoils

akabcenell, 2018-02-10:
So this is actually a design choice. The settings in the instruments as given in the Scripts tab are not automatically uploaded to the instrument, but must be done so in the Script. This is to prevent, for example, the standard setting for the piezo causing the objective to move every time a script with the piezo is run. However, it means that there are a bunch of fields in the Script tab which don't actually do anything. As a short term solution, I will go through and add this to the scripts that need it (for example, Daq_read_counter). Long term, I think it makes more sense to eliminate the settings tree in each instrument in the script tab. One will then only update the instrument settings through the instrument tab unless a specific setting is important for the script, in which case the script will have an appropriate setting and it will update the instrument. This avoids the case where in some scripts we want to make changes to the instrument in the script tab, in some we make changes in the instrument tab, and in some both work to varying degrees.

JanGieseler, 2018-05-01:
this behavior is a design choice. close issue

closed 2018-05-01 by JanGieseler