LISE-B26/pylabcontrol #128: Script loading with long filenames

JanGieseler, 2018-03-26:
There seem to be still some issues with loading long filenames. Maybe that is also due to the fact that the names have dots in them

e.g. in 
Z:\Lab\Cantilever\Measurements\20180322_Sample_37_diamond_Y_new_pulseblaster_script\180324-18_52_04_xy8_sweep

can't load subscript with
```python
Script.load_data(folder)
```
where 
```python
folder = './180324-18_52_04_xy8_sweep\\data_subscripts\\180324-18_52_14_xy8_k_double_init_1.25MHz_-13dBm_pi_pulse_blocks_k_1.000e+00'
```

asafira, 2018-03-26:
https://stackoverflow.com/questions/18390341/unable-to-locate-files-with-long-names-on-windows-with-python

akabcenell, 2018-03-26:
At some point, the following line was commented out in Script.load_data;

if len(path.split('\\\\?\\')) == 1:
    path = '\\\\?\\' + os.path.abspath(path)

This meant that the required \\\\?\\ was not appended to long paths and therefore they were not recognized appropriately. Uncommenting the line fixed this.

closed 2018-03-26 by akabcenell