LISE-B26/pylabcontrol #134: Gui: Saving fails with WindowsError: [Error 2] The system cannot find the file specified

akabcenell, 2018-04-02:
When saving to a good location (such as '\\\\?\\Z:/Lab/Lev/lab_measurements/20180323_sample_6/bead_1/cycle_3\\180402-19_41_11_takeimage'), saving fails in save_b26_file due to os.makedirs(os.path.dirname(filename)) failing on these filenames

akabcenell, 2018-04-03:
The solution was to remove the \\\\?\\ that was automatically appended to the filepath. This happened just after moving to pyqt5. Unclear why that would cause this though.

asafira, 2018-04-03:
I would prefer to close the issue if we can confirm that long save file names don't break...

closed 2018-06-25 by JanGieseler

akabcenell, 2018-06-25:
This should be fixed by Script.check_filename, which should be called before all automatic saving in Script. If are the further problems, reopen this issue

commit 06d3d374 (2018-06-25): Added a new function check_filename to script, which is called before saving or loading files in order to add the requisite '\\?\' prefix to filepaths that are longer than 256 characters on windows machines. If there are similar pathing issues on other os, they can be included in this function as well.
Closes #134