LISE-B26/pylabcontrol #135: get_config_value() in read_write_functions.py raises an error if no config file exists, should instead fail more gracefully

asafira, 2018-04-06:
Currently, if you try to load an instrument that requires a dll, but you don't have a config.txt file, we cannot import it and an exception is raised in get_config_value(). It would be great to fail more gracefully when we import those instruments, so that we can import them but can't create instances of them.

akabcenell, 2018-06-08:
todo: search for all config.txt and make sure that call to dll is inside the class definition

akabcenell, 2018-06-22:
All checks for config.txt should now occur within script class definitions, rather than at the top of the file outside of the class. This should solve this problem. Closing the issue for now, reopen it if this occurs in the future.

closed 2018-06-22 by akabcenell