LISE-B26/pylabcontrol #147: export default and loading scripts doesn't work on Eve

JanGieseler, 2018-06-13:
On the middle lab computer the scripts loading from the convert dialog doesn't work:
console output:
`bad magic number in 'b26_toolkit': b'\x03\xf3\r\n`
- checked the PYTHONPATH, which was not set correctly before (only pointing to pylabcontrol not to b26_toolkit and because of the all the instruments were broken, however after setting the PYTHONPATH this was not longer an issue. still loading scripts doesn't work

JanGieseler, 2018-06-13:
Console output (the issue might be the mix of / and \\ in the filepath)

classes_dict {}
files ['C:/Users/NV Experiment/PycharmProjects/pylabcontrol/pylabcontrol/scripts\\example_scripts.py', 'C:/Users/NV Experiment/PycharmProjects/pylabcontrol/pylabcontrol/scripts\\script_wait.py', 'C:/Users/NV Experiment/PycharmProjects/pylabcontrol/pylabcontrol/scripts\\select_points.py']
importing pylabcontrol.scripts.example_scripts
bad magic number in 'b26_toolkit': b'\x03\xf3\r\n'
importing pylabcontrol.scripts.script_wait
bad magic number in 'b26_toolkit': b'\x03\xf3\r\n'
importing pylabcontrol.scripts.select_points
bad magic number in 'b26_toolkit': b'\x03\xf3\r\n'
classes_dict {}

asafira, 2018-06-13:
I think Aaron probably just hasn't finished making the exporting work for packages not in the pythonpath. 

It might seem weird that it works on the other computers, but there is a setting in pycharm where we (temporarily) set pylabcontrol to depend on b26_toolkit, so that b26_toolkit is put into the pythonpath when we run anything in pylabcontrol.

akabcenell, 2018-06-13:
This seems to be an issue caused by old .pyc files not working when you switch from python 2->3, change folder structures, or bring them over from a different computer. It was solved by deleting all of the .pyc files in b26_toolkit and pylabcontrol and forcing python to create new ones from the current code.

closed 2018-06-13 by akabcenell

asafira, 2018-06-13:
Why would we be sharing.pyc files between computers? (I thought I deleted them all from GitHub a few months ago)

JanGieseler, 2018-06-13:
Did you test this on eve? When I checked there were not pyc files in the script folder...